chipsalliance/sv-tests #1227: Non-standard, but defacto standard tests

caryr, 2020-12-20:
While looking at more of the blacklisted ivtest tests I am finding that some of these indeed do not match the standard, but are allowed in Icarus because this functionality is support by commercial simulators. Part of our effort has been to be compatible with other simulators and the standard as long as the commercial extensions make sense and do not break compatibility with the standard.

I assume the test suite would be most useful to users if there was a way to designate these common commercial extensions and if they are supported.

For example VCS supports pull gates with multiple terminals. Not sure about any others.

We have comments in the code that NC-verilog supports => between multiple single bit values (e.g. (clk,d => q) = (tR_clk_q,tF_clk_q);) where clk, d and q are all single bit. VCS also supports this functionality

So how do you want to support this kind of behaviour? For now I will just add some documentation.

tgorochowik, 2020-12-21:
CC @hzeller

caryr, 2020-12-21:
I'll add '{} as another invalid syntax that is accepted by commercial tools (verified with VCS). It as expected represents an empty array.