chipsalliance/sv-tests #1241: CI runs are taking significantly longer to complete

caryr, 2020-12-30:
It looks like the install phase for each of the tools is taking significantly longer. This is almost doubling the time for some of the tests and is a huge waste of resources. For example many of the test were previously installed in about 5 minutes, but they are now in the half an hour range.

caryr, 2020-12-30:
Just to put this in perspective, this is over five hours of compute time per CI run so for the typical pull and then integrate that's over ten hours per pull request.

mithro, 2020-12-30:
I think this is likely to be because https://github.com/SymbiFlow/sv-tests/issues/1179

caryr, 2020-12-30:
I don't remember when this started being slower, but I think it was after #1179. I could be related to my cloning with a depth of 1 which is related to #1179. Though I think this was still okay after I pushed my first change.

caryr, 2020-12-30:
FYI #1226 was the last PR to run/be integrated with the shorter install times so that does point at the depth of 1 clone as the likely culprit. Not sure what needs to be done to fix this and allow shallow clones. What is odd is I did do an earlier commit, but that may have been before I initialized the submodules so maybe that's a clue. Looks like a git magician is needed here :-).

mithro, 2020-12-31:
After merging #1242 and removing the gh-pages branch from this repository. This bug should hopefully be fixed.

mithro, 2021-01-01:
Hopefully the CI run at https://github.com/SymbiFlow/sv-tests/actions/runs/455335796 will show this is now fixed...

caryr, 2021-01-01:
I took a quick look at the link and the Install phase is still around half an hour.

mithro, 2021-01-01:
@caryr You appear to be correct. I took a closer look at the run time and it seems to be mostly caused by the [`add-local-submodule.sh`](https://github.com/SymbiFlow/sv-tests/blob/master/.github/add-local-submodules.sh) script doing the checkout of all the submodules under `./third_party/cores/black-parrot/external/` particularly `./third_party/cores/black-parrot/external/riscv-gnu-toolchain`.

tgorochowik, 2021-01-04:
@mithro I think we could also consider getting rid of all the tool repos from the `third_party` dir. Originally we used that to build all the tools before testing and later kept them to make it possible to run everything locally without conda.

However I don't think anybody uses that anymore. It is much simpler to use the conda packages in the CI and locally everybody seems to be interested in testing a single tool from their own sources (which is understandable)

tgorochowik, 2021-01-04:
In addition the tool submodules seem confuse people about which version to expect in the report (#1181 #1208) which may be another reason to just remove them.

caryr, 2021-01-04:
As long as there is an easy way to manually trigger an update and whatever you do does not significantly increase the clone time I'm good. FYI I would have expected whatever was in third party to act as a seed for updating what you actually use which does seem to be the case for items in tests, but does not appear to to be the same for tools. It is this inconsistency that creates confusion. If you remove tools, please add a README explaining how to trigger a manual update.

Also if I am readying this correct it is cloning of the tests/core that is taking too much time not tools.

mithro, 2021-01-05:
@caryr - Yes, it is basically the cloning of git repositories which is taking a majority of the time...

See https://github.com/SymbiFlow/sv-tests/runs/1643846607?check_suite_focus=true for example...
 * Install - 48 minutes
 * Run - 11 minutes

mithro, 2021-01-05:
I'm unsure why the install time has gone back up to 50m verse the 20m...

 - https://github.com/SymbiFlow/sv-tests/runs/1635684226
 - https://github.com/SymbiFlow/sv-tests/runs/1643846607

tgorochowik, 2022-01-26:
The build times are improved significantly after switching to custom runners, let's close this

closed 2022-01-26 by tgorochowik