chipsalliance/sv-tests #1267: The iverilog that generates dashboard appears to be out of date

caryr, 2021-01-10:
There are 18 tests in the ivtest group that are unexpectedly failing when run with iverilog. All of them are related to recent fixes so it appears that the actual iverilog used to generate the dashboard is out of date even though the version linked in the repo has been updated a few times. There are also other general tests that should be passing based on recent fixes in iverilog. I was hoping whatever was being done for #1208 would also fix iverilog, but that was not the case.

tgorochowik, 2021-01-12:
Hi @caryr 

If you'd prefer to report the version of Icarus differently (or to add more details to it) please feel free to contribute a change to the runner:
https://github.com/SymbiFlow/sv-tests/blob/1bb97c0defa0fb64c90ddd6f6cf195d2b7944c88/tools/runners/Icarus.py#L43-L49

If you have a way to get for example a commit hash from which Icarus was built or something else that would make things simpler for you please don't hesitate to add that to the report.

As for what is used to run the tests and generate the reports: it is **not** the version from the submodule in the `third_party` dir. We are building a conda package for all the tools and simply download them to run the tests here. The packages are currently built here: https://github.com/symbiflow/conda-packages
but we intend to start using the packages from litex-hub very soon (https://github.com/litex-hub/)

caryr, 2021-02-08:
I have started a pull request that now gets the full version for Icarus.

Also the version of Icarus used to generate the dashboard has been updated, so many more tests are now passing. It is still somewhat out of date, but certainly better than before.

tgorochowik, 2021-02-08:
@caryr thanks for the PR

The version for icarus has indeed been frozen, see: https://github.com/litex-hub/litex-conda-eda/commit/400ad251da274dc645e0c603055042596fb4f808

The issue is that apparently the newer packages do not build anymore in ubuntu (which is used to build all conda packages).
The version freeze is a temporary workaround to be able to proceed with moving the packages to litex-hub as this blocked several other issues too.

Has anything changed with your build system or some dependencies lately? Would you be willing to change the line linked above back to master, creating a PR and investigating what causes the build to fail?

@MikePopoloski apparently slang has been experiencing a similar issue and its version is temporarily frozen too, see: https://github.com/litex-hub/litex-conda-eda/commit/5915d8faaa5a1a5a403d9f589f41f210254a9854

Our intent is to keep the packages up to date as much as possible, but we can't promise any timelines for that, so any help maintaining the packages for your tools would be greatly appreciated

CC @ajelinski

MikePopoloski, 2021-02-08:
The slang build failure looks like a bug in GCC 8. Any chance you can build with a newer version? My CI builds run with GCC 10 without issue.

ajelinski, 2021-02-08:
@MikePopoloski GCC 10 isn't available through Conda yet. Specifically, there's such a problem:
```
include/slang/binding/Lookup.h:217:26: note:   no known conversion for argument 2 from ‘const slang::SyntaxList<slang::ElementSelectSyntax>’ to ‘nonstd::span_lite::span<const slang::ElementSelectSyntax* const>’
  make[2]: *** [source/CMakeFiles/slangcompiler.dir/build.make:186: source/CMakeFiles/slangcompiler.dir/binding/Lookup.cpp.o]
```

Is there really no chance to fix it so that it can be built with older GCC?

MikePopoloski, 2021-02-08:
I can probably work around the specific issue (and I will try to do so), but it will be hard to not inadvertently break it again because I don't regularly build with old versions of GCC. GCC 11 will be shipping very soon and I will likely upgrade to that to start taking advantage of C++20 features.

Why is Conda so far behind? Surely there are other software packages out there that don't want to work around compiler bugs that were fixed years ago?

caryr, 2021-02-08:
From the parents failure log it's not obvious why this is failing. For Ubuntu we build using 18.04 and 20.04 in our CI and I personally build using 16.04 regularly. I know there was a small issue with configure compatibility that was resolved last month. Maybe that is the issue you were seeing. I did not actually make the change. @martinwhitaker committed the change a day after this was moved to a fixed version so maybe this has been resolved in the latest master.

It would really help if issues like this were reported in our git tracker so we can make sure they get addressed in a timely manner. We make an effort to address issue like this as quickly as possible.

caryr, 2021-02-09:
I will create a clone, make the change and create a PR.

caryr, 2021-02-09:
Well Linux is still failing (https://github.com/litex-hub/litex-conda-eda/pull/77). Both OSX and Windows built correctly. Which is odd for us since those two are the ones that usually create issues. How do I get access to the raw config.log referenced in the online logs to see what is actually breaking. The overall log file does not have enough information.

A couple oddities: I'm not sure what is up with all the placehold_ in the path, but that appears to not be an issue for the other tools or hte OSX build. Also the build is being tagged with v0_8_... which does not match what I have on my development machine though it does reference the latest commits made to the devel branch so maybe this is nothing. Maybe Martin has some insight.

tgorochowik, 2021-02-09:
The build script is located here:
https://github.com/litex-hub/litex-conda-eda/blob/8b2c216b4e9f27e890e42032443833796d743c04/sim/icarus/build.sh#L9

You could try temporarily changing the line to display the log if it fails, for example to:
```
./configure --prefix=$PREFIX || ( cat config.log ; false )
```

The `false` is there so the compilation is not continued if this step fails, feel free to experiment with the CI to get all the information that is needed.

Thank you for looking into this!

ajelinski, 2021-02-09:
@MikePopoloski TBH I have no idea why it's so far behind. Maybe it'll change now as only yesterday I saw GCC update (to 9.3) on anaconda.org (although it's not available through Conda yet). I saw some changes in how the compiler packages are divided now so it seems they're about to make a bigger change. It may mean new versions will be available earlier, but I don't know.

@caryr `placehold_` is a Conda oddity. `v0_8_` is used because it's the latest version-like tag on the `master` branch. The tags not resembling versions aren't used to set package version.

About the problem, I've checked some logs and options thoroughly and the problem seems to be with the commit changing configure scripts to work with autoconf 2.70, which is also not available on Conda yet (2.69 is used). The changes create a weird situation when the `./configure` says:
```
  checking for readline in -lreadline... yes
```
but when it checks if compiler works the `libreadline.so` can't be found:
```
  configure:8197: checking whether the C compiler works
  configure:8219: $BUILD_PREFIX/bin/x86_64-conda_cos6-linux-gnu-cc -g   conftest.c -lm -lpthread -lreadline  >&5
  $BUILD_PREFIX/bin/../lib/gcc/x86_64-conda_cos6-linux-gnu/7.3.0/../../../../x86_64-conda_cos6-linux-gnu/bin/ld: cannot find -lreadline
  collect2: error: ld returned 1 exit status
  configure:8223: $? = 1
  configure:8261: result: no
```

Maybe the `LDFLAGS` etc. environment variables are now overwritten by the `autoconf`? These variables are crucial to the way how Conda influences builds to use the Conda-based tools, libraries etc. Compiler packages, for example, set these variables among others:
```
  +CC=$BUILD_PREFIX/bin/x86_64-conda_cos6-linux-gnu-cc
  +CFLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem $PREFIX/include -fdebug-prefix-map=$SRC_DIR=/usr/local/src/conda/iverilog-v0_8_5903_g5530fbac -fdebug-prefix-map=$PREFIX=/usr/local/src/conda-prefix
  +CPPFLAGS=-DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -isystem $PREFIX/include
  +CPP=$BUILD_PREFIX/bin/x86_64-conda_cos6-linux-gnu-cpp
  +GCC=$BUILD_PREFIX/bin/x86_64-conda_cos6-linux-gnu-gcc
  +LDFLAGS=-Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,-rpath,$PREFIX/lib -Wl,-rpath-link,$PREFIX/lib -L$PREFIX/lib
```
so especially without `-L$PREFIX/lib` it simply won't work. On macOS I think it works without issues simply because the host machine has many things preinstalled and Linux machines aren't so packed in CI.

MikePopoloski, 2021-02-09:
My latest commit gets slang building with GCC 8 again: https://github.com/MikePopoloski/slang/commit/8c79db3b019d67c2c317d15a007a8c707222784f

Can you update litex hub?

ajelinski, 2021-02-09:
@MikePopoloski Oh, great! I'm testing if it can be built successfully now. I'll merge it to master if there are no issues, so the packages will be available on LiteX-Hub's Anaconda channel in a few hours.

caryr, 2021-02-09:
Thank you Adam for the detailed explanation and Mike for the suggestion. I will try to do some research/experimentation later.

caryr, 2021-02-09:
Sorry, that should have been Tomasz!. Though I do appreciate Mike for other contributions :).

caryr, 2021-02-09:
One further question. Is there an option to only run Icarus to limit resource usage during debugging? I assume I could hack all the config files to fail early, but that seems heavy handed.

ajelinski, 2021-02-09:
@caryr You can remove all jobs apart from `slang-linux` in `.github/workflows/build-packages.yml`. :) There's no need to change `NUM_OF_JOBS` if you remove the `master-package` job.

ajelinski, 2021-02-09:
@caryr I meant "apart from `icarus-(linux|osx|windows)`" of course!

ajelinski, 2021-02-09:
@MikePopoloski The `slang-v0.6_35_g8c79db3b` package can be now downloaded from the LiteX-Hub channel. :)

MikePopoloski, 2021-02-09:
Great! Do you know when the sv-tests dashboard will update with the newer version?

tgorochowik, 2021-02-10:
The report is updated each time something is pushed to the master branch. We have a few outstanding PRs so it should be soon.

I guess adding a job to rebuild the report daily even if nothing changes in the sources is worth considering though. CC @hzeller

caryr, 2021-02-14:
I tried updating to the latest Icarus in litex-hub and it failed. I have looked at the changes to configure and I see nothing that should have caused this. The issue we were resolving related to newer version of autoconf not supporting the AC_PROC_CC_C99 macro since C99 is enabled automatically by default. It should have had nothing to do with readline!

I have read more about conda than I every expected to and could find nothing that points at the issue either. The only thing I could think of, but it does not explain why things worked previously is does the host readline get exported correctly for building? We did update the configure to have the .ac suffix instead of .in so no idea if that causes the configure script to be built differently and maybe more thoroughly check for readline. I'm not sure what else I can do to resolve this.

tgorochowik, 2022-01-26:
Everything is now built from submodules and should be up to date, conda is no longer required. Closing

closed 2022-01-26 by tgorochowik