chipsalliance/sv-tests #1287: yosys is listed as a runner for all tests

caryr, 2021-01-20:
I was looking at the iverilog run and noticed yosys was also listed as a runner and it along with iverilog was running all the test. I checked the other tests and it looks like all of them (at least all that I checked) are also running yosys so that is causing a lot of extra redundant testing.

tgorochowik, 2021-01-20:
Thanks for the report!

It seems to be because we always install yosys plugins which depend on yosys - the CI then discovers then yosys is installed and uses it in all tests.

Hopefully this will be fixed with #1274 as one of the things it does is it fixes the dependencies in yosys conda packages

tgorochowik, 2021-03-11:
This seems to be already fixed.

closed 2021-03-11 by tgorochowik