chipsalliance/sv-tests #1348: SweRV core has possible syntax errors

MikePopoloski, 2021-02-16:
It looks like the core has legitimate syntax errors where code refers to variables before they are declared. Example:
```
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec_tlu_ctl.sv:410:34: error: identifier 'lsu_error_pkt_dc4' used before its declaration
   rvoclkhdr lsu_e4_e5_cgc ( .en(lsu_error_pkt_dc4.exc_valid | lsu_exc_valid_wb | clk_override), .l1clk(lsu_e4_e5_clk), .* );
                                 ^~~~~~~~~~~~~~~~~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec_tlu_ctl.sv:748:20: note: declared here
   lsu_error_pkt_t lsu_error_pkt_dc4;
                   ^
```
Not sure if you would get that fixed upstream or what.

tgorochowik, 2021-02-17:
It is indeed present upstream (i.e. this is not generated code):

Usage in line 413:
https://github.com/chipsalliance/Cores-SweRV/blob/ec254f54915fbd38d1878fca9247bec12db2fcaf/design/dec/dec_tlu_ctl.sv#L413

and definition in line 758:
https://github.com/chipsalliance/Cores-SweRV/blob/ec254f54915fbd38d1878fca9247bec12db2fcaf/design/dec/dec_tlu_ctl.sv#L758

Is it strictly forbidden by the LRM? I am asking just because other tools seem to allow that, I am not sure whether they should. If not we should probably file an issue in upstream

MikePopoloski, 2021-02-17:
The LRM seems pretty clear to me, but it's true that most commercial tools only loosely enforce the rule, which is very confusing. Some example excerpts from the LRM:
From 3.12.1 (Compilation units):
```
task t;
int x;
x = 5 + b; // illegal - "b" is defined later
x = 5 + $unit::b; // illegal - $unit adds no special forward referencing
endtask
bit b;
```

From 6.5 (Nets and Variables):
> Data shall be declared before they are used, apart from implicit nets 

From 6.21 (Scope and lifetime):
> A variable declaration shall precede any simple reference (non-hierarchical) to that variable.

Also this part from 6.10 (Implicit declarations):
> If an identifier is used in the terminal list of a primitive instance or in the port connection list of a
module, interface, program, or static checker instance (but not a procedural checker instance, see
17.3), and that identifier has not been declared previously in the scope where the instantiation
appears or in any scope whose declarations can be directly referenced from the scope where the
instantiation appears (see 23.9), then an implicit scalar net of default net type shall be assumed

It's actually extra weird because each commercial tool I tried was fine with a *port* referencing declarations later in scope but not any other construct, including continuous assignments which have the exact same ruleset for implicit nets. For example, this doesn't error:
```
module n(input int a); endmodule

module m;
  n n1(.*);
  int a;
endmodule
```

But this *does* error, even though they follow the exact same rules for implicit nets!
```
module m;
  assign a = 1;
  int a;
endmodule
```
Unless I'm missing something in the LRM, I would say that these tools are just buggy, or maybe more likely they were originally shipped with bugs and are now stuck supporting them for backwards compatibility purposes.

MikePopoloski, 2021-02-17:
Ah, of course wildcard and implicit-named port connections don't create implicit nets, which is why the rules differ. However, yet another place in the LRM makes it clear this is still broken.

From 23.3.2.3 (Connecting module instance using implicit named port connections (.name)):
> It shall be an error if the name port_identifier has not been declared (explicitly or implicitly) or
imported from a package (by explicit or wildcard import) prior to the .port_identifier implicit port
connection

tgorochowik, 2021-02-17:
Nice, thanks for the detailed analysis! I have already created a PR for that as you've probably noticed

MikePopoloski, 2021-02-17:
Ah yes. To be clear, the error I posted was only one example. The full list is:
```
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/ifu/ifu.sv:262:21: error: identifier 'ifu_bp_kill_next_f2' used before its declaration
   ifu_ifc_ctl ifc (.*
                    ^~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/ifu/ifu.sv:271:11: note: declared here
   logic  ifu_bp_kill_next_f2; // kill next fetch; taken target found
          ^
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/ifu/ifu.sv:283:19: error: identifier 'ifc_fetch_req_f1' used before its declaration
   ifu_bp_ctl bp (.*);
                  ^~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/ifu/ifu.sv:289:32: note: declared here
   logic ifc_fetch_req_f1_raw, ifc_fetch_req_f1, ifc_fetch_req_f2;
                               ^
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec_tlu_ctl.sv:409:64: error: identifier 'lsu_error_pkt_dc4' used before its declaration
   rvoclkhdr lsu_e3_e4_cgc ( .en(lsu_error_pkt_dc3.exc_valid | lsu_error_pkt_dc4.exc_valid | lsu_error_pkt_dc3.single_ecc_error | lsu_error_pkt_dc4.single_ecc_error | clk_override), .l1clk(lsu_e3_e4_clk), .* );
                                                               ^~~~~~~~~~~~~~~~~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec_tlu_ctl.sv:748:20: note: declared here
   lsu_error_pkt_t lsu_error_pkt_dc4;
                   ^
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec_tlu_ctl.sv:409:131: error: identifier 'lsu_error_pkt_dc4' used before its declaration
   rvoclkhdr lsu_e3_e4_cgc ( .en(lsu_error_pkt_dc3.exc_valid | lsu_error_pkt_dc4.exc_valid | lsu_error_pkt_dc3.single_ecc_error | lsu_error_pkt_dc4.single_ecc_error | clk_override), .l1clk(lsu_e3_e4_clk), .* );
                                                                                                                                  ^~~~~~~~~~~~~~~~~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec_tlu_ctl.sv:748:20: note: declared here
   lsu_error_pkt_t lsu_error_pkt_dc4;
                   ^
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec_tlu_ctl.sv:410:34: error: identifier 'lsu_error_pkt_dc4' used before its declaration
   rvoclkhdr lsu_e4_e5_cgc ( .en(lsu_error_pkt_dc4.exc_valid | lsu_exc_valid_wb | clk_override), .l1clk(lsu_e4_e5_clk), .* );
                                 ^~~~~~~~~~~~~~~~~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec_tlu_ctl.sv:748:20: note: declared here
   lsu_error_pkt_t lsu_error_pkt_dc4;
                   ^
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec.sv:512:25: error: identifier 'dec_i0_cinst_d' used before its declaration
   dec_ib_ctl instbuff (.*
                        ^~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec.sv:551:17: note: declared here
   logic [15:0] dec_i0_cinst_d;
                ^
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec.sv:515:27: error: identifier 'dec_i0_cinst_d' used before its declaration
   dec_decode_ctl decode (.*);
                          ^~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec.sv:551:17: note: declared here
   logic [15:0] dec_i0_cinst_d;
                ^
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec.sv:517:21: error: identifier 'dec_tlu_i0_valid_wb1' used before its declaration
   dec_tlu_ctl tlu (.*);
                    ^~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/dec/dec.sv:557:32: note: declared here
   logic dec_tlu_i1_valid_wb1, dec_tlu_i0_valid_wb1,  dec_tlu_int_valid_wb1;
                               ^
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/swerv.sv:993:13: error: identifier 'mexintpend' used before its declaration
            .*
            ^~
../../src/github/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/src/chipsalliance.org_cores_SweRV_EH1_1.8/design/swerv.sv:1011:17: note: declared here
   logic        mexintpend;
                ^
```

tgorochowik, 2021-02-17:
Right, I should not have tested it so selectively (I just checked that single module).

I updated the PR and tested the whole core with slang

MikePopoloski, 2021-03-27:
Looks like the fix has finally propagated into sv-tests, so closing this out.

closed 2021-03-27 by MikePopoloski