chipsalliance/sv-tests #1400: Missing define found in verilog config *.vlt in Ibex trips some parsers

hzeller, 2021-03-17:
Parsers that actually can deal with *.vlt files that is.

There is a *.vlt file in Ibex that expands a `verilator_config` [1] macro that it doesn't see. So either this is something generated from the wrong kind of fusesoc setting, or we need to include the define beforehand.

Example: 
 * sv2v https://symbiflow.github.io/sv-tests-results/#Sv2v_zachjs|ibex|ibex
  * sv-parser https://symbiflow.github.io/sv-tests-results/#sv_parser|ibex|ibex

Interestingly, it works in Verilator: https://symbiflow.github.io/sv-tests-results/#Verilator|ibex|ibex  -- maybe this is just a predefined value there ?

[1]: interesting meta question: how to put a backtick in a back-ticked region in github markdown ?

MikePopoloski, 2021-03-17:
Surely verilator-specific files should only be passed to verilator and not other tools? I raised a similar issue in #1347

hzeller, 2021-03-17:
I'll assign both related ones to Tom to have a look.

tgorochowik, 2021-03-24:
> [1]: interesting meta question: how to put a backtick in a back-ticked region in github markdown ?

I think the easiest way is to use multiple `` ` `` to escape it, for example
``` `` ` `` ``` produces `` ` `` 
```` ``` `` ` `` ``` ```` produces ``` `` ` `` ``` 
etc

MikePopoloski, 2021-06-03:
I fixed this in #1545

closed 2021-06-04 by tgorochowik