chipsalliance/sv-tests #1437: fix the variances in column width in the report

tgorochowik, 2021-04-07:
Currently all the columns have different widths, it does not look very good:
[image]

It would be nice to make all the widths equal. We need to think about what can  be done so the headers are still equally readable.

Maybe we could make them vertical? Or use some kind of an angle? Maybe it could use multiple lines and wrap? We need to experiment with that and see what works the best.

hzeller, 2021-04-07:
I was wondering about that as well. I played with a 45 degree angle once (there are some [CSS hacks for that](https://css-tricks.com/rotated-table-column-headers/)), and it wasn't looking too bad and it can make things much more compact.

Prahitha, 2021-04-11:
I've tried a couple of options, please let me know which one would suit best. Thanks!

[image]
[image]
[image]
https://user-images.githubusercontent.com/44160152/114308437-a62a6180-9b01-11eb-9ab8-8113a72c560f.mp4

tgorochowik, 2021-04-12:
Those look really nice! Thank you for your work!

My vote goes for the 45 angle one. In the final version we will have to make sure the "sort" icon is aligned to the end of the name nicely (currently it seems to be at a constant position and is covered by the text in some cases).

What do you think @hzeller ?

Prahitha, 2021-04-12:
Thank you for the feedback! Also, I was wondering if it would be okay to split the word (for the wrap option), i.e, replace the underscore with a space):
`<a class="tool_link" target=_blank href='{{ report[tool]["url"] }}'>{{ tool.lower() }}</a>` -> `<a class="tool_link" target=_blank href='{{ report[tool]["url"] }}'>{{ tool.replace("_", " ").lower() }}</a>`
I think this will make the word-break look cleaner.

hzeller, 2021-04-13:
+1 to the 45 degree version. That looks very nice and will always work; which is harder for the wrap version to make look pretty in all circumstances.

The up/down sorting arrow thing might be something that I'd intuitively expect not to be 'up in the air' next to the text, but more  aligned with the top of the table, upright; a bit like this:

[image]

Hopefully that can be implemented without too much of a CSS scramble

tgorochowik, 2021-04-30:
This is already implemented, closing.

closed 2021-04-30 by tgorochowik