chipsalliance/sv-tests #1462: Ibex test targets Xilinx-specific wrapper

MikePopoloski, 2021-04-18:
The Ibex synthesis test is targeting the top level wrapper that includes Xilinx-specific module instantiations. No tool other than Vivado is going to be able to build this, so probably the test should be just for the core itself instead of the artya wrapper (indeed, no tools pass this test except the ones that don't do any semantic checking).

MikePopoloski, 2021-05-02:
Bump! Anyone who knows the fusesoc setup better that can adjust the build? I'm pretty sure it's incorrect right now.

mithro, 2021-05-02:
@olofk - Could you help here?

kgugala, 2021-05-03:
@MikePopoloski I don't agree - Yosys/UHDM can synthesize this design. Pure Yosys fails, but not on missing Xilinx primitives, but rather on unsupported SV language constructs.
In general, for synthesis, we need to target some FPGA device. Xilinx is as good as any other.

MikePopoloski, 2021-05-03:
The clkgen_xil7series.sv file instantiates modules like "IBUF" and "PLLE2_ADV" which are not declared anywhere in the project. They are not defined anywhere in the LRM. All commercial simulators I tried will error if you try to instantiate one of those modules. This repo is for testing general SystemVerilog language support of tools; what is the purpose of including a core that can only pass with one specific tool for which it is specialized, which can't even be simulated, only synthesized?

kgugala, 2021-05-03:
synthesis tools have those defined internally so they understand what those blocks are.

For simulation we use `ibex_simple_system` target https://github.com/SymbiFlow/sv-tests/blob/master/conf/fusesoc-configs/ibex-sim.yml#L15 (this one does not refer to Xilinx specific stuff).

MikePopoloski, 2021-05-03:
Ok, so the problem is the compatible runners then? Not sure why verilator is the only one specified for the simulator, and the synthesis target is being given to all other tools when yosys is the only synthesizer?

kgugala, 2021-05-03:
If it is like that, it need to be fixed. I'd set synth target only of synthesis tools and sim for all the rest

olofk, 2021-05-05:
Well, I agree that it should use a synthesis wrapper that doesn't use vendor-specific primitives. That sounds like a bad idea. Why not just make a vendor-independent wrapper without these things?

kgugala, 2021-05-05:
There are two types of tests here - SV spec unit tests and real-life projects tests. Ibex is in the latter group. The idea with those is that we take the designs as they are (without adapting them to pass the tests).

Ibex has Xilinx synthesis target, and Vendor independent simulation target. All the synth tools we test in the suite can handle Xilinx synth target. The problem here (as @MikePopoloski wrote) is that we use this synth target to test non-synthesis tools. We're working on fixing that - to use vendor independent simulation target to test e.g. parsers or formatters.

olofk, 2021-05-05:
Got it. Still think it would make sense to have a vendor-neutral synthesis wrapper, but that should be added on the LowRISC side rather than this test suite then

MikePopoloski, 2021-05-21:
Any progress on this?

tgorochowik, 2021-05-21:
It turned out that we cannot simply convert the remaining runners to use the simulation target as then they would get the Verilator-specific files (`*.vlt` ) on input.

We could in theory use ` --PRIM_DEFAULT_IMPL=prim_pkg::ImplGeneric` (and make sure all tools get this define) - this is a generic synthesis target without the Xilix-specific stuff.

Most of the source code in Ibex has a check for this and uses vendor independent code for this, however it looks as though there is not vendor independent top module so it's not easy to generate the whole core to be a test without some extra work.

MikePopoloski, 2021-05-21:
Can we instead add a filter on the vlt files and only pass them to verilator?

MikePopoloski, 2021-06-03:
I ended up fixing this in #1545

closed 2021-06-03 by MikePopoloski