chipsalliance/sv-tests #1463: Taiga core is completely busted

MikePopoloski, 2021-04-19:
There are many issues with the taiga core:
1. Many instances of variables being used before they're declared, for example:
```
../../src/github/sv-tests/third_party/cores/taiga/core/instruction_metadata_and_id_management.sv:207:35: error: identifier 'id_not_in_decode_issue' used before its declaration
        .toggle((gc_init_clear & ~id_not_in_decode_issue) | (decode_advance & ~gc_fetch_flush)),
                                  ^~~~~~~~~~~~~~~~~~~~~~
../../src/github/sv-tests/third_party/cores/taiga/core/instruction_metadata_and_id_management.sv:295:11: note: declared here
    logic id_not_in_decode_issue;
          ^
```

2. The xilinx_byte_enable_ram module is not included in the project; probably missing the core/xilinx folder in the generator?
3. mstatus_priv_reg.sv references numerous enums members that don't exist. It looks like these were maybe renamed and nobody bothered to update this one file? For example:
```
../../src/github/sv-tests/third_party/cores/taiga/core/mstatus_priv_reg.sv:53:62: error: use of undeclared identifier 'SUPERVISOR_PRIV'; did you mean 'SUPERVISOR_PRIVILEGE'?
    assign exception_privilege_level = exception_delegated ? SUPERVISOR_PRIV : MACHINE_PRIV;
                                                             ^~~~~~~~~~~~~~~
../../src/github/sv-tests/third_party/cores/taiga/core/csr_types.sv:32:9: note: declared here
        SUPERVISOR_PRIVILEGE = 2'b01,
        ^
```

4. id_inuse.sv has undeclared identifiers that appear to simply not exist anywhere in the source code...
5. div_unit_core_wrapper.sv instantiates the div_algorithm module and provides parameters, even though div_algorithm doesn't have any parameters defined!
6. l2_fifo.sv references 'l2_fifo_interface' which doesn't appear anywhere in the source code...
7. taiga_wrapper instantiates the `taiga` module with a `.*` but does not provide a connection for one of the ports, which is illegal.

At this point I gave up trying to make it work. Probably this should just be removed from the test suite?

MikePopoloski, 2021-05-02:
Bump on this? Any thoughts at all?

mithro, 2021-05-04:
@kgugala / @hzeller - Can you make sure someone takes a look at this? Thanks!

MikePopoloski, 2021-05-21:
Bumping again! If nobody cares about this should I put up a PR to remove it?

tgorochowik, 2021-05-21:
@MikePopoloski I think we can remove it and for now just create a new issue to restore it in a proper way to restore it with correct configs. Would you be willing to do that?

MikePopoloski, 2021-05-21:
Yeah, I'll take a look at a PR