chipsalliance/sv-tests #1469: Keep the first column (tags) fixed on scroll-x

Prahitha, 2021-04-21:
Similar to the fixed header, having the first column fixed would make it easier to browse the logs/report

Prahitha, 2021-04-25:
So here I tried replacing the fixedHeader extension with fixedColumns. The one with no margin looks kind of close to what is expected but there is some overflow in the header, so I tried adding a margin and overflow: hidden but that clearly did not work. I would like some suggestion/views on this, thank you!

Without margin:
https://user-images.githubusercontent.com/44160152/116006884-c7736d80-a62a-11eb-89bb-055e37de42d2.mp4

With margin:
https://user-images.githubusercontent.com/44160152/116006823-8a0ee000-a62a-11eb-97ce-02eafa3de765.mp4

tgorochowik, 2021-04-26:
The first one looks very promising!

Could you please create a PR with the changes? It will be easier to discuss this if we can look at the code

PR #1484 (2021-05-19): fixed first column of the table

fixes #1469 and #1470
closed 2021-05-19 by tgorochowik