chipsalliance/sv-tests #1501: Build fails with "No rule to make target 'image/bin/surelog'"

udif, 2021-05-06:
I'm trying to build c8f960aad61ef55f7b5056cab90550f983eb1c77 under Ubuntu 20.04 inside WSL2+Windows 10, and whatever I try, I end up with:

```
  Build successful.

make[1]: Leaving directory '/home/udif/git/sv-tests/third_party/cores/ibex-yosys-build/yosys'
install -D /home/udif/git/sv-tests/third_party/tools/../cores/ibex-yosys-build/yosys/yosys /home/udif/git/sv-tests/out/runners/bin/antmicro-yosys
mkdir -p /home/udif/git/sv-tests/out/runners/bin/
cd /home/udif/git/sv-tests/third_party/tools/uhdm-integration && make image/bin/surelog
make[1]: Entering directory '/home/udif/git/sv-tests/third_party/tools/uhdm-integration'
make[1]: *** No rule to make target 'image/bin/surelog'.
make[1]: Leaving directory '/home/udif/git/sv-tests/third_party/tools/uhdm-integration'
make: *** [tools/runners.mk:92: uhdm-common] Error 2
make: Target 'runners' not remade because of errors.
```
I started with the instructions on the main page (git clone, submodules update, pip3 install) and then installed along the way any missing package I've encounered (haskell stack, uuid-dev, cargo, etc.), and eventually ended with the message above.
(Along the way I had to trim the PATH from Windows executables added by WSL2 that contained spaces in their names).

I've been trying everything I can think of:
* removing old binaries (yosys, verilator, etc.) in /usr/local that might interfere with sv-tests supplies binaries
* removing -j flag for a single threaded (and anguishingly long) build
* removing -k flag so it doesn't try to keep building anything else if it fails
* `make clean` and retrying (on any of the stages above)
* In addition to `make clean` above I was also looking inside the `third party` submodules, and cleaning any local file created there.
* Eventually, reinstalling **everything** from scratch (rm -rf sv-tests and do the whole thing again) just in case a missing package or library along the way caused some autoconf or makefiles to be created incorrectly, **and** ran it single threaded again.

I still the same result.

I see that the relevant line in runners.mk was added a few months ago by @hzeller on 5273e464162cbcbe18568e06b54740a7e2882f1c .
I'm embarrassed to say I don't know how to debug this (I can't find the `image/bin/surelog` rule in `tools/runners.mk`).
However, `out/runners/bin` seems tocontain all the executables already built.
```
~/git/sv-tests$ ls out/runners/bin/
antmicro-yosys                 verible-transform-interactive.sh                     verilator_bin
git-verible-verilog-format.sh  verible-verilog-diff                                 verilator_bin_dbg
iverilog                       verible-verilog-format                               verilator_coverage
iverilog-vpi                   verible-verilog-format-changed-lines-interactive.sh  verilator_coverage_bin_dbg
moore                          verible-verilog-kythe-extractor                      verilator_gantt
odin_II                        verible-verilog-lint                                 verilator_profcfunc
parse_sv                       verible-verilog-obfuscate                            vvp
slang-driver                   verible-verilog-preprocessor                         yosys
surelog                        verible-verilog-project                              zachjs-sv2v
uhdm-dump                      verible-verilog-syntax
verible-patch-tool             verilator
```
I also have a full , 566K, in-order (no parallel make)  build log (no 'make -j') that can be useful for debugging this.

Any help would be appreciated!

tgorochowik, 2021-05-07:
Hi @udif 

the Makefile is not directly used in sv-tests to build the tools, instead we use tools installed from the conda repos ( https://anaconda.org/LiteX-Hub/repo).

It is possible that the Makefile here has bitrotted a bit, let's continue the discussion in https://github.com/alainmarcel/uhdm-integration/issues/321 first, and then we can attempt to fix the Makefile here.

zekungao, 2021-07-02:
@udif 
If you are in hurry, may be you can temporarily modify the `runner.mk` as I did in my personal fork:7ec92dc3c3247ced327d7ab812f15be117e841fd

tgorochowik, 2022-01-04:
The makefiles are now fixed and used in the CI, closing. Thanks!

closed 2022-01-04 by tgorochowik