chipsalliance/sv-tests #1533: How to add UVM based VIP

taichi-ishitani, 2021-05-18:
Hi,

I' developing an UVM based AMBA AXI VIP named tvip-axi and I'd like to add my VIP to the tests.
https://github.com/taichi-ishitani/tvip-axi

How can I get source files for uvm_pkg?
I need those source files to compile my VIP.

tgorochowik, 2021-05-20:
Hi

we provide uvm as a submodule in the third-party directory, see: https://github.com/SymbiFlow/sv-tests/tree/master/third_party/tests

You can take a look at the existing UVM tests to see how they use it, for example: https://symbiflow.github.io/sv-tests-results/#Slang|uvm-agents|uvm_agent_active

There are some tests that are defined directly in the tests directory, for example: https://github.com/SymbiFlow/sv-tests/blob/master/tests/chapter-16/16.12--property-uvm.sv

and some that use a generator, for example: https://github.com/SymbiFlow/sv-tests/blob/master/conf/generators/templates/uvm-classes_0.sv and https://github.com/SymbiFlow/sv-tests/blob/master/conf/generators/templates/uvm-classes_0.json

Feel free to reuse stuff from those tests

taichi-ishitani, 2021-05-20:
Hi @tgorochowik ,
Thank you for your reply! I understood.

closed 2021-05-20 by taichi-ishitani