chipsalliance/sv-tests #1551: TNoC errors

tgorochowik, 2021-05-26:
It looks as though the TNoC module that has been added some time ago produces errors in almost all of the tools.

For example:
Slang:
```
../../src/github/sv-tests/tests/generated/tnoc/tnoc.sv:28:33: error: interface port 'types' not connected
  tnoc_flit_if #(PACKET_CONFIG) receiver_if[9]();
                                ^
../../src/github/sv-tests/third_party/cores/tnoc/rtl/common/tnoc_flit_if.sv:10:15: note: declared here
  tnoc_types  types
              ^
../../src/github/sv-tests/tests/generated/tnoc/tnoc.sv:29:33: error: interface port 'types' not connected
  tnoc_flit_if #(PACKET_CONFIG) sender_if[9]();
                                ^
../../src/github/sv-tests/third_party/cores/tnoc/rtl/common/tnoc_flit_if.sv:10:15: note: declared here
  tnoc_types  types
              ^
../../src/github/sv-tests/tests/generated/tnoc/tnoc.sv:31:3: error: unknown module 'tnoc_fablic'
  tnoc_fablic #(PACKET_CONFIG) u_fablic (.*);
  ^~~~~~~~~~~
Top level design units:
    tnoc
```

Surelog:
```
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[SNT:PA0207] third_party/cores/tbcm/tbcm_crc.sv:2:2: Syntax error: no viable alternative at input 'interface tbcm_crc\n  import',
  import  tbcm_crc_pkg::*;
  ^-- ./slpp_all/work/tbcm_111217407228608498/tbcm_crc.sv:2:2:.

[SNT:PA0207] third_party/cores/tnoc/rtl/common/tnoc_types.sv:4:2: Syntax error: no viable alternative at input 'interface tnoc_types\n  import',
  import  tnoc_pkg::*;
  ^-- ./slpp_all/work/common_6660747120724478387/tnoc_types.sv:4:2:.

[SNT:PA0207] third_party/cores/tnoc/rtl/common/tnoc_utils.sv:2:2: Syntax error: no viable alternative at input 'interface tnoc_utils\n  import',
  import  tnoc_pkg::*;
  ^-- ./slpp_all/work/common_6660747120724478387/tnoc_utils.sv:2:2:.

[SNT:PA0207] third_party/cores/tnoc/rtl/common/tnoc_flit_if.sv:4:2: Syntax error: no viable alternative at input 'interface tnoc_flit_if\n  import',
  import  tnoc_pkg::*;
  ^-- ./slpp_all/work/common_6660747120724478387/tnoc_flit_if.sv:4:2:.

[SNT:PA0207] third_party/cores/tnoc/rtl/common/tnoc_packet_if.sv:4:2: Syntax error: no viable alternative at input 'interface tnoc_packet_if\n  import',
  import  tnoc_pkg::*;
  ^-- ./slpp_all/work/common_6660747120724478387/tnoc_packet_if.sv:4:2:.

[SNT:PA0207] third_party/cores/tnoc/rtl/common/tnoc_address_decoder_if.sv:4:2: Syntax error: no viable alternative at input 'interface tnoc_address_decoder_if\n  import',
  import  tnoc_pkg::*;
  ^-- ./slpp_all/work/common_6660747120724478387/tnoc_address_decoder_if.sv:4:2:.

[SNT:PA0207] third_party/cores/tnoc/rtl/router/tnoc_port_control_if.sv:4:2: Syntax error: no viable alternative at input 'interface tnoc_port_control_if\n  import',
  import  tnoc_pkg::*;
  ^-- ./slpp_all/work/router_5987062843406567056/tnoc_port_control_if.sv:4:2:.
```

It looks as though some files may still be missing.

@taichi-ishitani could you please take a look at the report to see what is missing? Almost all of the tools fail with it, there are probably some issues with its configuration (missing files? incorrect order?) Thanks!

taichi-ishitani, 2021-05-26:
Hi,

I've fixed these errors and open PR #1552 .

tgorochowik, 2021-05-26:
Nice! Thank you for a very quick response

PR #1552 (2021-05-26): fix TNoC wrapper error

This PR is to fix #1551.
I've confirmed there are no errors by using VCS.
closed 2021-05-26 by tgorochowik