chipsalliance/sv-tests #1553: Error when downloading the tnoc submodule

Prahitha, 2021-05-27:
[image]

taichi-ishitani, 2021-05-27:
Hi @tgorochowik 
I often edit repositories added as submodules directly so I prefer `git@` protocol to `https`.

mithro, 2021-05-27:
By default you should use https and then use `insteadOf` to change what is used https://gist.github.com/Kovrinic/ea5e7123ab5c97d451804ea222ecd78a / https://git-scm.com/docs/git-config

taichi-ishitani, 2021-05-28:
Hi,
I've changed TNoC's submodule config according to advice from @mithro .

taichi-ishitani, 2021-05-29:
> We can close this issue after updating the submodule

Should I create a PR to update TNoC?

tgorochowik, 2021-05-31:
Please do! (dependabot should have updated it but it seems it didn't)

taichi-ishitani, 2021-05-31:
I've created PR #1567 for this issue.

PR #1567 (2021-06-01): update TNoC

Update TNoC to fix #1553 .
closed 2021-06-01 by tgorochowik