chipsalliance/sv-tests #1686: Problems When Module Parameters are Passed Single Words of a Multi Bit Array in SystemVerilog but not Verilog

WhiteNinjaZ, 2021-08-11:
This issue was originally posted as part of [symbiflow-examples issue #170](https://github.com/SymbiFlow/symbiflow-examples/issues/170)

@hzeller  I have been testing several third party designs with symbiflow/Yosys and have encountered a rather perplexing bug. While working on the [Ad Astra](https://projectf.io/posts/fpga-ad-astra/) designs from project F, I encountered a module that generated a faulty bit-stream when written in SystemVerilog but not when written in Verilog.  It appears this issue is caused by symbiflow's inability to handle passing specific words of a multi bit array into a module. To demonstrate this, I have included a design that passes each word of a 3x4-bit wire `multiBit` between two modules:
 
```
module top(
    input logic [13:0] sw,
    output logic [3:0] led
    );
    logic [3:0] multiBit [0:2];
    
    test m1(sw[3:0], multiBit[0]);
    test m2(sw[7:4], multiBit[1]);
    test m3(sw[11:8], multiBit[2]);

    assign led =
    (sw[13:12] == 0) ? multiBit[0]:
    (sw[13:12] == 1) ? multiBit[1]:
    multiBit[2];

endmodule

```

```
module test(
    input logic [3:0] switch_val,
    output logic [3:0] multiB
    );

    assign multiB = switch_val;

endmodule

```
This design is meant to display the value of 4 switches (sw[3:0], sw[7:4], or sw[11:8]) on the first 4 LEDs of a basys board. The user can choose which switch set to display on the LEDs by toggling switches 12 and 13 on and off (i.e switch 12 high and switch 13 low will display the value of switches 4-7 on the LEDs). Notice that each word of the multi bit array multiBit is passed into the module named `test`. The `test` module simply assigns the value of a 4 bit input to the value of a 4 bit output. 

When I run the above modules through symbiflow, the toolchain generates a faulty bitstream. If I convert the file to Verilog, as demonstrated bellow, the design works properly: 
```
module top(
    input wire [13:0] sw,
    output wire [3:0] led
    );
    wire [3:0] multiBit [0:2];
    
    test m1(sw[3:0], multiBit[0]);
    test m2(sw[7:4], multiBit[1]);
    test m3(sw[11:8], multiBit[2]);

    assign led =
    (sw[13:12] == 0) ? multiBit[0]:
    (sw[13:12] == 1) ? multiBit[1]:
    multiBit[2];

endmodule
```
```
module test(
    input wire [3:0] switch_val,
    output wire [3:0] multiB
    );

    assign multiB = switch_val;

endmodule
```

I have tested a few variations of the SV design, and it appears that symbiflow can handle accessing individual words of a multi dimensional array in a SystemVerilog design so long as the array is only accessed and modified within a single module. Passing 1D arrays of bits between separate modules also works without flaw. From what I can see, symbiflow only breaks on SV designs when specific words of a 2D array are passed into another module as a parameter. 

I believe this problem might be coming from the Yosys part of the symbiflow tool-chain. 

What is the best way to proceed?

WhiteNinjaZ, 2021-08-17:
@hzeller?

mithro, 2021-08-17:
@kgugala - You were talking about this recently IIRC?

hzeller, 2021-08-17:
I was a bit confused as why this is in sv-tests, but looking at https://github.com/SymbiFlow/symbiflow-examples/issues/170 this makes more sense.

This is not an issue we can _solve_ in sv-tests, but we can document it here - sv-tests collects all kinds of examples and regressions to check if they parse with various tools; so when Tim mentioned to get it into sv-tests, that is what he meant with that. So for this project here: adding this example to the set of sv-tests we have can be useful (though sv-tests mostly checks if the code can be parsed, and does not verify if the bitstream is correct). So in this thread we should discuss how to reduce the test-case to put into sv-tests.

As for Yosys  - there is a known issue where multi-dimensional arrays don't exactly work with Yosys; @kgugala 's team is working on these; not sure if this is the particular issue here. So with regard to this question, https://github.com/SymbiFlow/symbiflow-examples/issues/170 might be the more relevant project to continue the discussion.