chipsalliance/sv-tests #1744: Tools not being updated again

MikePopoloski, 2021-09-07:
Tool dependencies seem to be not getting updated again. The generated dashboard seems to have tool versions that are about a month old for several that I checked (slang, Surelog, verible). I've lost track of where the actual versions are coming from these days; conda, GitHub actions, submodules via dependabot? Regardless, it would be nice to see up-to-date tool versions used.

mithro, 2021-09-07:
We really need something which alerts if these tools are not being updated...

mithro, 2021-09-07:
@hzeller - Who should own this?

hzeller, 2021-09-07:
This should be owend by @tgorochowik 
I suspect it also depends on work by @QuantamHD to get cloud runner machines available.

mithro, 2021-09-13:
Can we set up some type of alerting so that we know when this happens?

MikePopoloski, 2021-09-15:
Looks like the builds are failing again? At least from what I can tell here: https://github.com/hdl/conda-eda/actions

kamilrakoczy, 2021-09-17:
The builds should be fixed after: https://github.com/hdl/conda-eda/pull/131

alaindargelas, 2021-11-25:
At least Surelog is not being updated anymore, I don't know about the other tools.

tgorochowik, 2021-11-25:
@alaindargelas as we discussed offline, we are back to using submodules to generate the report. This way there is more control over what is built when. The submodules are bumped weekly, if you need more frequent updates, feel free to bump the submodule more often.

Thanks!

closed 2021-11-25 by tgorochowik

alaindargelas, 2021-11-26:
I'm saying the Surelog submodule has not been updated in over 2 weeks. I did not nor intend to trigger an update manually.
So something is broken in the automatic update.
Surelog version should read 1.14, it is stuck at 1.13.