chipsalliance/sv-tests #1759: uhdm-common fails to build

toddstrader, 2021-09-16:
I'm getting this when running `make -k runners`:
```
cd /usr/scratch/local/devs/sv-tests/third_party/tools/uhdm-integration && make image/bin/surelog
/bin/sh: line 0: cd: /usr/scratch/local/devs/sv-tests/third_party/tools/uhdm-integration: No such file or directory
tools/runners.mk:91: recipe for target 'uhdm-common' failed
make: *** [uhdm-common] Error 1
make: Target 'runners' not remade because of errors.
```

There is no submodule at `third_party/tools/uhdm-integration`.  Is that missing?  I thought maybe it was this:
https://github.com/chipsalliance/UHDM
but that doesn't seem to have a make target called `image/bin/surelog`.

Please advise.

hzeller, 2021-09-16:
This looks to have happened here:
https://github.com/chipsalliance/sv-tests/commit/01de648edaa93c491a86673c91ddfffae0f3da60 (PR #1702)

I suspect since now yosys is using the UHDM module that was deemed obsolete, but that rule still needs it to build the

Adding @kamilrakoczy to fix and @tgorochowik as he is working on running the makefile directly in the custom runners CI not pulling a conda build.

hzeller, 2021-09-16:
@toddstrader short term workaround: if you 
```
make -k runners
```

the remaining targets should be built (`-k` stands for *keep going*).

tgorochowik, 2021-12-03:
All the runners are now built in the CI so it should all be fixed. Please let us know if you find any new issues. Thanks!

closed 2021-12-03 by tgorochowik