chipsalliance/sv-tests #1817: How to rerun one test for one tool (Using the whole test harness)

alaindargelas, 2021-10-17:


mglb, 2021-10-17:
`make RUNNERS_FILTER=Verible tests`, or just `make RUNNERS_FILTER=Verible`
You might need to remove existing test results for the tool: `rm -r out/logs/Verible`

(replace `Verible` with a tool name).

alaindargelas, 2021-10-30:
That will run all tests, how about one test?

tgorochowik, 2021-11-02:
First remove the output file for which you want to re-run the test, e.g:
```
rm out//logs/Verilator/chapter-10/10.3.1--net-decl-assignment.sv.log
```
and then run:
```
make out//logs/Verilator/chapter-10/10.3.1--net-decl-assignment.sv.log
```

Note the double `//` after out

alaindargelas, 2021-11-06:
@tgorochowik, @hzeller, After poking around for some time, that is the culprit:
https://github.com/chipsalliance/sv-tests/blob/d442e183aa409005f1181b0d11bd7b214a421a6e/tools/BaseRunner.py#L23

Surelog is performing full elaboration and dumping the UHDM datamodel, so it does not fit in 4Gb for the large tests. They are all failing in sv-tests since the elaboration was turned on.

How do we go about it?

closed 2021-11-06 by alaindargelas