chipsalliance/sv-tests #1839: How to change the max memory limit for certain tests, and globably for a runner

alaindargelas, 2021-11-02:
@tgorochowik, Surelog is hitting a limit (3Gb?) set somewhere. Modifying the top Makefile does not change the limit.

tgorochowik, 2021-11-03:
This is the only place where it is set:
https://github.com/chipsalliance/sv-tests/blob/d442e183aa409005f1181b0d11bd7b214a421a6e/Makefile#L20

BTW I don't think this mechanism is used at the moment (it's disabled) so you shouldn't be hitting the limit.

You can also see that in plots uploaded as artifacts for each job:
[image]

Also note the CPU usage for surelog - it's similar for the uhdm clients here in sv-tests.

alaindargelas, 2021-11-04:
I ran Valgrind + Surelog in the context of sv-test make , Valgrind tells me that Surelog hit a memory limit set by the system and exits prematurely. This does not happen if I run external to sv-test make on the same testcase. So something somewhere in sv-tests runner system is setting a limit on memory usage

@hzeller, do you know?

alaindargelas, 2021-11-06:
@tgorochowik, @hzeller, After poking around for some time, that is the culprit:
sv-tests/tools/BaseRunner.py

Line 23 in d442e18

 resource.setrlimit(resource.RLIMIT_DATA, (4 << 30, 4 << 30)) 
Surelog is performing full elaboration and dumping the UHDM datamodel, so it does not fit in 4Gb for the large tests. They are all failing in sv-tests since the elaboration was turned on.

How do we go about it?

tgorochowik, 2021-11-08:
Indeed! I was not aware of that.

Let's try to allow the tools to use more memory (we cant do that with the custom runners) and see where it gets us: https://github.com/chipsalliance/sv-tests/pull/1845

closed 2021-11-16 by alaindargelas