chipsalliance/sv-tests #1867: Earlgrey design is not complete

alaindargelas, 2021-11-22:
Surelog complains about missing packages,
if I compare the Surelog Earlgrey design, it contains those packaes in under autogen directories, in sv-tests, those autogen directories do not exist.
One example of such missing package in the sv-test env:

alain@alain-xps:~/alain-marcel/sv-tests/build$ grep -r "package" . | grep flash_ctrl_pkg

alain@alain-xps:~/Surelog/third_party/tests/Earlgrey_nexysvideo$ grep -r "package" . | grep flash_ctrl_pkg
./src/lowrisc_systems_flash_ctrl_pkg_0.1/rtl/autogen/flash_ctrl_pkg.sv:package flash_ctrl_pkg;
./src/lowrisc_systems_flash_ctrl_pkg_0.1/rtl/autogen/flash_ctrl_pkg.sv:endpackage : flash_ctrl_pkg

MikePopoloski, 2021-11-22:
I opened #1564 a while ago for this same reason.

alaindargelas, 2021-11-24:
Thanks for connecting the dots in between the Issues, I'll discuss this with the team.

RRozak, 2022-01-25:
Fixed by https://github.com/chipsalliance/sv-tests/pull/2116

closed 2022-01-25 by RRozak