chipsalliance/sv-tests #2169: SweRV core is broken

MikePopoloski, 2022-01-29:
The core is trying to include files that don't exist in the design. For example:
```
error: no such file or directory: '/root/sv-tests/sv-tests/build/chipsalliance.org_cores_SweRV_EH1_1.8/sim-verilator/../src/chipsalliance.org_cores_SweRV_EH1_1.8/testbench/ahb_sif.sv'
```

RRozak, 2022-01-31:
This error is nondeterministic.
Today the SweRV target failed on Verilator: https://github.com/chipsalliance/sv-tests-results/blob/25b9e8ece69978580778222cfd19f6a676d8abb1/logs/Verilator/generated/fusesoc/swerv-sim.sv.log.html
yesterday it passed: https://github.com/chipsalliance/sv-tests-results/blob/70339344708d3de0a3eee1ca12ddd2887eae24f0/logs/Verilator/generated/fusesoc/swerv-sim.sv.log.html
and there was no commits between them.

I think that the problem is similar to the one with ibex-sim in https://github.com/chipsalliance/sv-tests/pull/2148, because there is also no file and I had problems with reproducing the error. Since I've reproduced it already, I will continue investigation on that target.

RRozak, 2022-02-03:
This problems doesn't seem to be similar to the one from ibex. 
I think that the problems occur, because swerv-sim and swerv-synth are installed to the same directory and one override files of another. I changed it: https://github.com/chipsalliance/sv-tests/pull/2197 and it seems to work.

closed 2022-02-07 by RRozak