chipsalliance/sv-tests #2314: Coverage system functions 20.14 is illegal verilog

alaindargelas, 2022-02-23:
initial begin
        i = $coverage_control(`SV_COV_CHECK, `SV_COV_TOGGLE, `SV_COV_HIER, $root);
                                                                                                                                               ^------ $root cannot be used without a hierarchical path, no commercial tool accepts this.

MikePopoloski, 2022-02-28:
The LRM says it's allowed. See section 40.3.2.1, they give that exact line as an example.

alaindargelas, 2022-03-02:
All the commercial tools fail with a syntax error and the grammar does not support a naked $root. 
https://insights.sigasi.com/tech/systemverilog.ebnf/#hierarchical_identifier
How do you reconciled that?

MikePopoloski, 2022-03-02:
Yeah, it's weird for sure, but this is certainly not the first legit feature in the LRM that commercial tools don't support, and it's also not the first feature to be described concretely in prose but not be represented correctly in the grammar. The LRM is simply not very rigorous about consistency.

I don't feel like the coverage control example is a typo because of the description: "checks all instances to verify they have coverage" which you couldn't accomplish without having syntax like this, but I also don't really care too much as it's likely nobody in the world has ever tried to do this (since most tools don't support it anyway).

alaindargelas, 2022-03-06:
OK, so how did you alter your grammar to support the naked $root? Did you change the hierarchical_identifier? Or something more targeted?

MikePopoloski, 2022-03-06:
I allow the standalone $root during parsing and then check at analysis time whether it should be allowed (which only the coverage control system tasks do). It wasn't too difficult with my setup: https://github.com/MikePopoloski/slang/commit/81c0103ad1d9d92e525c6a5499ba8c5b843e29c3

alaindargelas, 2022-03-10:
Got it.

closed 2022-03-10 by alaindargelas