chipsalliance/sv-tests #2522: Tests for proper simulation semantics?

Kuree, 2022-03-31:
Most tests here seem to focus on parsing/elaboration. Is it possible to extend it to test the simulation semantics as well? For instance, is the sensitivity list deduction done correctly or is the event control implemented properly. Most tests in `ivtest` follow PASS/FAIL printout checker, which might be a good starting point.

I'm asking this because I'm working on a SystemVerilog [simulator](https://github.com/Kuree/fsim) and would love to have a test suite to work with.

Thanks.

mithro, 2022-03-31:
I think we would probably **love** to have that but we have never had the time / resources to work on it.

 @hzeller can provide more guidance.

tgorochowik, 2022-03-31:
We do have a way to test simulation actually, please take a look at this:
https://github.com/chipsalliance/sv-tests/blob/master/tools/runners/Verilator.py#L40

So for simulation, Verilator gets a different set of arguments.

And the tests look like this: https://github.com/chipsalliance/sv-tests/blob/3589dc3804673208c6edbd9577e856516b087f78/tests/chapter-11/11.3.6--assign_in_expr-sim.sv

Note the:
```
    $display(":assert: (5 == %d)", a);
    $display(":assert: (5 == %d)", b);
    $display(":assert: (5 == %d)", c);
```

The output of simulation tests is parsed, we look for assert blocks and evaluate the expressions with python.

This is somewhat limited, but allows us to run some basic, one-fits-all simulation testing.

Feel free to submit extensions if you have any ideas! You can also contribute additional tests or add your simulator as an extra runner if you want to.

Kuree, 2022-04-06:
Thanks. I will take a look and reopen this issue if I encounter any problem.

closed 2022-04-06 by Kuree