chipsalliance/sv-tests #2749: tools/sv-report very slow

hzeller, 2022-05-23:
It takes about 40 minutes CPU time for the sv-report tool to run and creating a report (And it runs in a single thread).
This is painfully slow.
Are there some simple things that can be done to speed it up ?

tgorochowik, 2022-11-03:
```
   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.051    0.051   63.582   63.582 sv-report:1(<module>)
      142    0.000    0.000    0.000    0.000 sv-report:120(escapeXmlAttribute)
        2    0.000    0.000    0.000    0.000 sv-report:124(escapeJsString)
       86    0.000    0.000    0.000    0.000 sv-report:131(surroundWithQuotes)
     3289    0.002    0.000    0.010    0.000 sv-report:136(numericSortKey)
     2275    0.001    0.000    0.002    0.000 sv-report:138(<lambda>)
       64    0.000    0.000    0.000    0.000 sv-report:142(numericSort)
        5    0.000    0.000    0.000    0.000 sv-report:146(numericDictSort)
       11    0.000    0.000    0.000    0.000 sv-report:147(<lambda>)
        3    0.000    0.000    0.000    0.000 sv-report:152(tagDictSort)
       42    0.000    0.000    0.000    0.000 sv-report:153(tagSortKey)
       42    0.000    0.000    0.000    0.000 sv-report:159(<lambda>)
     8741    0.008    0.000    0.076    0.000 sv-report:173(exists_and_is_newer_than)
     3009    0.005    0.000    0.033    0.000 sv-report:185(totalSize)
        2    0.001    0.000    0.002    0.001 sv-report:200(readConfig)
        1    0.000    0.000    0.000    0.000 sv-report:235(TestStatus)
     3051    0.002    0.000    0.004    0.000 sv-report:242(__str__)
        1    0.000    0.000    0.000    0.000 sv-report:246(TestResult)
        1    0.000    0.000    0.000    0.000 sv-report:272(TagInfo)
        1    0.000    0.000    0.000    0.000 sv-report:278(ToolInfo)
        1    0.000    0.000    0.000    0.000 sv-report:288(TagToolResult)
      280    0.000    0.000    0.000    0.000 sv-report:295(total_tests)
      126    0.000    0.000    0.000    0.000 sv-report:299(status)
        1    0.000    0.000    0.000    0.000 sv-report:310(ToolSummary)
        1    0.000    0.000    0.000    0.000 sv-report:329(GroupData)
        1    0.000    0.000    0.000    0.000 sv-report:342(PartialGroupData)
        1    0.000    0.000    0.000    0.000 sv-report:350(ReportResults)
     3009    0.003    0.000    2.834    0.001 sv-report:380(convertPathsToRelativeLinks)
    46498    0.181    0.000    1.945    0.000 sv-report:396(convertPath)
     3009    0.049    0.000    4.194    0.001 sv-report:423(renderLogToHTMLFile)
     5732    0.106    0.000   57.907    0.010 sv-report:456(renderInputFileToHTMLFile)  <<<<
       42    0.004    0.000    0.046    0.001 sv-report:485(renderTagResultsConfig)
        2    0.258    0.129    5.317    2.658 sv-report:511(collect_logs)
        2    0.000    0.000    0.000    0.000 sv-report:512(LocalPartialGroupData)
     3009    0.008    0.000    0.292    0.000 sv-report:602(<listcomp>)
     3108    0.001    0.000    0.011    0.000 sv-report:698(<lambda>)
       42    0.000    0.000    0.000    0.000 sv-report:709(<listcomp>)
        2    0.000    0.000    0.000    0.000 sv-report:735(<dictcomp>)
        1    0.000    0.000    0.000    0.000 sv-report:769(<listcomp>)
```

Here is some profiling output (the relevant part, and only for a few tests to make it quicker to run/compare).

The least we can do is generate the logs in a few parallel jobs to speed this up. I will submit a PR.

tgorochowik, 2022-11-03:
Fixed in #3498

closed 2022-11-03 by tgorochowik