chipsalliance/sv-tests #4751: 11.4.8: continuous assignments races with initial block

caryr, 2023-07-19:
In the 11.4.8 section there are tests like the following:

```
module top();
wire [31:0] a;
wire [31:0] b;
wire [31:0] c;
assign a = 32'd4;
assign b = 32'd3;
assign c = a & b;
initial begin
    $display(":assert: (int(%s) == %d)", "4&3", c);
end
endmodule
```

Icarus does not correctly report the result since the `initial` block is executing before the three continuous assignments have completed. Adding any delay, even an explicit zero delay `#0` will resolve the race and allows Icarus to report the correct results. I could not find anywhere in the standard that required continuous assignments to execute before an `initial` block so I believe these tests should be updated, but I would also be interested in a reference to a location in the standard that mandates this behavior.

I did test this using one of the commercial simulators and it did work correctly, but that would also likely be correct behavior since I expect any order of execution would be allowed even one that consistently allowed the continuous assignments to run first. The standard is explicit about `initial` and `always` being any order and in section 4.3 it does define continuous assignments as processes that are concurrently scheduled like `initial` and `always` blocks so I would assume the standard does not specify an execution order.

kbieganski, 2023-07-21:
As you said, IEEE defines continuous assignments as processes, and it explicitly says that there is no implied ordering between `always` and `intial` processes. Unfortunately, it seems to omit continuous assignments from that consideration.

However, based on IEEE 4.8 I think it's safe to say that your conclusion about the test being ambiguous is correct. Adding a delay does not seem right though, as that would complicate the test and require timing control support.

Perhaps it'd be better to turn these continuous assignments into statements in the `initial` block. The template for these tests can be found [here](https://github.com/chipsalliance/sv-tests/blob/469266729e1761ed2cec7dfcfdce0e56535ae27a/conf/generators/templates/operators-sim.json).

Would you be willing to submit a PR with a fix?

caryr, 2023-07-21:
You could also use a final block since that has implicit delay.

In Icarus the continuous assignments and the procedural assignments actually use different code to implement the expressions so ideally you would have both for full coverage.

I'll see if I can find some time to make the appropriate changes.

kbieganski, 2023-07-21:
I thought about `final` but I believe that breaks Yosys and Odin. Though they may already be broken on these tests

caryr, 2023-07-21:
Okay, then what about `$strobe()` instead of `$display()`? That also resolves the race since `$strobe()` is defined to run at the end of the current simulation step instead of immediately.

kbieganski, 2023-07-24:
Good idea. We can try and see if it breaks anything in a PR.

closed 2023-07-24 by tgorochowik