chipsalliance/sv-tests #4776: Backout RgGen?

wsnyder, 2023-07-25:
#1534 merged in adding RgGen.  (@tgorochowik @taichi-ishitani)

While I appreciate adding more examples this added code fails on every tool supported.  As general policy I'd suggest sv-tests not allow merging new code unless: it either passes on every tool, or explains for each fail that the failure is an issue with the tool.  (As opposed to a configuration problem in sv-tests, which is what is wrong here.)

I'd like to see this change backed out until the code is resubmitted with proper configuration for the tools.

Why does this matter?  I don't think the tool developer's should have to debug new submissions to make them work - this should be the job of the submitter.

Thanks for considering.

MikePopoloski, 2023-07-25:
FWIW I agree with this; it's very annoying when nobody bothers to make sure new tests work with even one single tool. Whether intended or not, it comes across as pure laziness.

I don't know that you have to provide explanations for literally every tool that fails; sv-tests has a number of tools that have very minimal support for the language and/or fail on a huge number of tests, but for those that pass most tests it seems reasonable to check newly added failures to see if they're the fault of the new test or the tool itself.

taichi-ishitani, 2023-07-25:
https://github.com/chipsalliance/sv-tests/blob/572e3913b8fc45cc1172c6b3827f9c280caf3b4b/generators/rggen#L27

The top module is based on the previous sample RgGen module but the sample module and dependent moodules are the latest.
I think it causes the error.
I will fix the top module.

tgorochowik, 2023-07-25:
This is my fault. I misinterpreted the test changes table due to broken headers seen in https://github.com/chipsalliance/sv-tests/pull/1534#issuecomment-1620037553 (the headers have been fixed since then), and then failed to check the actual results due to the limited time I can devote to sv-tests.

#4777 reverts this, @taichi-ishitani feel free to contribute your change, either as a fix or as a new PR if #4777 finishes more quickly.

Thanks and sorry about this

closed 2023-07-25 by tgorochowik

MikePopoloski, 2023-07-25:
Nice, thanks for the quick response!

taichi-ishitani, 2023-07-26:
I've create PR #4778 to fix this error.
Could you please review this PR?

PR #4777 (2023-07-25): Revert #1534
Closes #4776
PR #4778 (2023-07-26): update RgGen sample
Hi,

I've update the top module for RgGen sample modules.
This is to fix #4776.
commit 3f5c65fe (2023-07-26): update RgGen sample to fix #4776