chipsalliance/sv-tests #5082: Yosys SV Support Misrepresented Due to $display

sifferman, 2023-10-01:
Hello, and thank you for assembling this very useful record of SV support in tools! It has been super interesting to follow.

## Problem

There are several tests that Yosys is incorrectly failing due to using $display. Yosys does not support the `$display` statement for non-constant values:

```plain
ERROR: Failed to evaluate system task `$display' with non-constant argument.
```

Many more tests would pass if their values were verified by other means than `$display`. To list a few:

* [tests/chapter-5/5.6.4--compiler-directives-preprocessor-macro_1.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-5/5.6.4--compiler-directives-preprocessor-macro_1.sv)
* [tests/chapter-7/arrays/packed/equality.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-7/arrays/packed/equality.sv)
* [tests/chapter-7/arrays/packed/onebit.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-7/arrays/packed/onebit.sv)
* [tests/chapter-7/memories/read-write.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-7/memories/read-write.sv)
* [tests/chapter-7/structures/packed/basic.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-7/structures/packed/basic.sv)
* [tests/chapter-7/structures/packed/signed.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-7/structures/packed/signed.sv)
* [tests/chapter-7/structures/packed/unsigned.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-7/structures/packed/unsigned.sv)
* [tests/chapter-7/unions/packed/basic.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-7/unions/packed/basic.sv)
* [tests/chapter-10/10.4.1--blocking-assignment.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-10/10.4.1--blocking-assignment.sv)

Many more tests *correctly* fail because the feature is unsupported, but will eventually incorrectly fail because they use `$display`. To list a few:

* [tests/chapter-7/arrays/multidimensional/copy.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-7/arrays/multidimensional/copy.sv)
* [tests/chapter-7/arrays/multidimensional/subarrays.sv](https://github.com/chipsalliance/sv-tests/tree/master/tests/chapter-7/arrays/multidimensional/subarrays.sv)

## Possible Solutions

To fix this, a custom assert macro/function should be added that would operate differently depending on the tool choice. Here are a couple possibilities for what the custom assert macro/function should do:

### `output ERROR` Port
Possibly all the tests mentioned could have an `output ERROR` port added to the module definition, and the assert function/macro could just run `ERROR |= !(<assertion>);` for Yosys runs. Then, [`eval`](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/eval.html) could be run to check the value of `ERROR`.

### External Script

Another possibility is that the custom assert macro/function does nothing when run by Yosys. Instead, an external script will parse all the asserts and run [`eval`](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/eval.html) on each one.

(Note that this will not work for the tests [onebit.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-7/arrays/packed/onebit.sv) and [read-write.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-7/memories/read-write.sv) because there are modifications to values after they have an assertion)

tgorochowik, 2023-10-09:
Hi, thank you for the report.

Let me start by saying that originally all the tests that use `display` with asserts inside were meant for simulator tests. Then we added "elaboration-only" class which made it possible to run those in other tools (like yosys). So we could also just disable those tests for yosys, the things they cover (besides the asserts meant for simulation) _should_ already be covered by other test cases.

That being said, I'm not opposed to adjusting those tests to make then usable for tools that only run elaboration. I'm not sure if having an error output port makes sense for yosys? would we be able to read the calculated value there?

One additional idea - maybe we could wrap those display/assert calls with some ifdefs (e.g. `ifdef SIMULATION`) and define this flag for all the simulators we test.

Feel free to propose a PR with a fix or a workaround for the problem.

sifferman, 2023-10-10:
> I'm not sure if having an error output port makes sense for yosys? would we be able to read the calculated value there?

Yes! Yosys' eval command calculates the values of all output ports of a specified module. If we add an `ERROR` port, then Yosys can calculate the value

I just made an example repo that demonstrates this: [sifferman/eval-example](https://github.com/sifferman/eval-example). I adjusted the tests mentioned above, then run synthesis and `eval -set-undef top`.

> we could also just disable those tests for yosys, the things they cover (besides the asserts meant for simulation) should already be covered by other test cases.

I guess I am also concerned about false positives if Yosys doesn't throw an error when parsing. Using `eval` can test whether the synthesis is correct.

sifferman, 2023-10-10:
I would be happy to make these changes if you think it sounds good

sifferman, 2023-10-10:
I spent some time sorting through the tests to determine which should be enabled/disabled for Yosys.

These should be **modified** for Yosys:

* [chapter-11/11.10--string_bit_array-sim.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-11/11.10--string_bit_array-sim.sv)
* [chapter-12/12.6.1--casex_pattern.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.6.1--casex_pattern.sv)
* [chapter-12/12.6.1--casez_pattern.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.6.1--casez_pattern.sv)
* [chapter-12/12.6.1--case_pattern.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.6.1--case_pattern.sv)
* [chapter-12/12.6.2--if_pattern.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.6.2--if_pattern.sv)
* [chapter-12/12.7.1--for.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.7.1--for.sv)
* [chapter-12/12.7.2--repeat.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.7.2--repeat.sv)
* [chapter-12/12.7.3--foreach.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.7.3--foreach.sv)
* [chapter-12/12.8--break.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.8--break.sv)
* [chapter-12/12.8--continue.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.8--continue.sv)
* [chapter-12/12.8--return.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.8--return.sv)
* [chapter-12/12.8--return_val.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-12/12.8--return_val.sv)
* [chapter-13/13.4--function.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-13/13.4--function.sv)
* [chapter-13/13.4.1--function-return-assignment.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-13/13.4.1--function-return-assignment.sv)
* [chapter-13/13.4.1--function-return.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-13/13.4.1--function-return.sv)
* [chapter-13/13.4.1--function-void-return.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-13/13.4.1--function-void-return.sv)
* [chapter-13/13.4.2--function-automatic.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-13/13.4.2--function-automatic.sv)
* [chapter-13/13.4.2--function-recursive.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-13/13.4.2--function-recursive.sv)
* [chapter-13/13.4.2--function-static.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-13/13.4.2--function-static.sv)
* [chapter-13/13.4.3--const-function.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-13/13.4.3--const-function.sv)
* [chapter-20/20.15--random.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.15--random.sv)
* [chapter-20/20.5--itor.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.5--itor.sv)
* [chapter-20/20.5--real-bits-conv.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.5--real-bits-conv.sv)
* [chapter-20/20.5--rtoi.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.5--rtoi.sv)
* [chapter-20/20.5--shortreal-bits-conv.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.5--shortreal-bits-conv.sv)
* [chapter-20/20.6--bits.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.6--bits.sv)
* [chapter-20/20.6--bits_type.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.6--bits_type.sv)
* [chapter-20/20.7--array-queries-multi-dim.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.7--array-queries-multi-dim.sv)
* [chapter-20/20.7--array-queries.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.7--array-queries.sv)
* [chapter-20/20.8--atan2.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--atan2.sv)
* [chapter-20/20.8--ceil.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--ceil.sv)
* [chapter-20/20.8--clog2.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--clog2.sv)
* [chapter-20/20.8--exp.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--exp.sv)
* [chapter-20/20.8--floor.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--floor.sv)
* [chapter-20/20.8--hypot.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--hypot.sv)
* [chapter-20/20.8--ln.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--ln.sv)
* [chapter-20/20.8--log10.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--log10.sv)
* [chapter-20/20.8--pow.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--pow.sv)
* [chapter-20/20.8--sqrt.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.8--sqrt.sv)
* [chapter-20/20.9--countbits.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.9--countbits.sv)
* [chapter-20/20.9--onehot.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-20/20.9--onehot.sv)
[... cut ...]
* [chapter-9/9.7--process_cls_suspend_resume.sv](https://github.com/chipsalliance/sv-tests/blob/master/tests/chapter-9/9.7--process_cls_suspend_resume.sv)

Note: it's a bit difficult what tests should be enabled/disabled for synthesis tools, because ideally they should be able to run all constant functions/constructs that don't depend on input ports. I will go back through this list and check how the big 3 handle ambiguous cases. Or is there a place to see how the big 3 respond to all tests in this repo? That could be a useful metric.

tgorochowik, 2023-10-10:
Thank you for the work. You suggestions sound good, just make sure the tests are not getting much more complicated than they are now (at least in terms of readability and number of different SV constructs).

> Or is there a place to see how the Synopsys/Cadence/Siemens tools respond to all tests in this repo? That could be a useful metric.

The licenses of the proprietary tool disallow using them to run in any benchmarks or comparisons so we cannot have this here.

If you do wish to compare (other) tools that allow it, take a look at how the runners in the `tools` directory are implemented. Adding your own tool should be fairly straightforward - so you should be able to run some comparisons locally

daglem, 2024-01-09:
I just added a simulation pass to the Yosys runner in #5498, in anticipation of merge of https://github.com/YosysHQ/yosys/pull/3963 in Yosys.
I have tested the Yosys PR, and it solves the current issues with `$display` in sv-tests, and gives a nice boost in passed tests for Yosys.
I believe this negates the need to change any tests.

daglem, 2024-01-17:
The recent boost in passed tests for Yosys is due to https://github.com/YosysHQ/yosys/pull/4129.

By default, output from `$display` et al. in `initial` blocks is produced both from elaboration and simulation. The tests still work, however duplication of the output is confusing, and could possibly cause problems later on. For simulation tests, #5525 suppresses the output from elaboration, leaving only output from simulation.