chipsalliance/sv-tests #5807: Ariane core test is still broken

MikePopoloski, 2024-07-05:
The recent change definitely helped things, but the build is still not configured correctly. Most relevant slang error:
```
../../../third_party/cores/ariane/core/cache_subsystem/cva6_hpdcache_subsystem.sv:171:12: error: 'hpdcache_typedef.svh': No such file or directory
  `include "hpdcache_typedef.svh"
           ^~~~~~~~~~~~~~~~~~~~~~
```

tgorochowik, 2024-07-08:
Thanks for checking!

@Moschn would you mind taking a further look?

Moschn, 2024-07-08:
Yeah, I just now saw the logs in the dashboard. I probably should set sv-tests up locally to fix see such issues before I see the logs in the online dashboard. Doing that now. 

In the meantime, I created a fix for this specific issue and opened a PR #5808.

tgorochowik, 2024-07-08:
Let's keep it open until we verify if the fix fixed everything indeed

Moschn, 2024-07-08:
I tried some tools locally now, and with the bump (#5809), the generated flow should be correct. Some tools still do not manage to parse/compile the core, but this is now a SV/source code issue.

For example, Slang reports:
```
../../../third_party/cores/ariane/corev_apu/tb/ariane_testharness.sv:752:26: error: no implicit conversion from 'int' to 'string'; explicit conversion exists, are you missing a cast?
        string verbosity = 0;
```

If this is fixed in source code by lets say `string verbosity;` then Slang passes aswell. I am not sure if setting a string to 0 is supported systemverilog, but it seems that it is accepted by some tools. If this is not officially supported in SV, tell me, then I can probably fix this in the core.

tgorochowik, 2024-07-08:
Thanks for checking!

Are you using the built-in runner? The report shows a different error:
```
slang-driver --single-unit --timescale=1ns/1ns --top=ariane_testharness -I /root/sv-tests/sv-tests/third_party/cores/ariane/core/include -I /root/sv-tests/sv-tests/third_party/cores/ariane/vendor/pulp-platform/common_cells/include -I /root/sv-tests/sv-tests/third_party/cores/ariane/vendor/pulp-platform/common_cells/src -I /root/sv-tests/sv-tests/third_party/cores/ariane/vendor/pulp-platform/axi/include -I /root/sv-tests/sv-tests/third_party/cores/ariane/common/local/util -I /root/sv-tests/sv-tests/third_party/cores/ariane/core/cache_subsystem/hpdcache/rtl/include -I /root/sv-tests/sv-tests/third_party/cores/ariane/corev_apu/axi_node -I /root/sv-tests/sv-tests/third_party/cores/ariane/vendor/pulp-platform/common_cells/include -I /root/sv-tests/sv-tests/third_party/cores/ariane/vendor/pulp-platform/axi/include -I /root/sv-tests/sv-tests/third_party/cores/ariane/corev_apu/register_interface/include -I /root/sv-tests/sv-tests/third_party/cores/ariane/corev_apu/tb/common -I /root/sv-tests/sv-tests/third_party/cores/ariane/vendor/pulp-platform/axi/include -I /root/sv-tests/sv-tests/third_party/cores/ariane/verif/core-v-verif/lib/uvm_agents/uvma_rvfi -I /root/sv-tests/sv-tests/third_party/cores/ariane/verif/core-v-verif/lib/uvm_agents/uvma_core_cntrl -I /root/sv-tests/sv-tests/third_party/cores/ariane/verif/tb/core -I /root/sv-tests/sv-tests/third_party/cores/ariane/core/include -Wno-error=index-oob -Wno-error=range-oob -Wno-error=range-width-oob -DVERILATOR [third_party/cores/ariane/vendor/pulp-platform/fpga-support/rtl/SyncDpRam.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/pulp-platform/fpga-support/rtl/SyncDpRam.sv.html) [third_party/cores/ariane/vendor/pulp-platform/fpga-support/rtl/AsyncDpRam.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/pulp-platform/fpga-support/rtl/AsyncDpRam.sv.html) [third_party/cores/ariane/vendor/pulp-platform/fpga-support/rtl/AsyncThreePortRam.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/pulp-platform/fpga-support/rtl/AsyncThreePortRam.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_pkg.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_pkg.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_cast_multi.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_cast_multi.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_classifier.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_classifier.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_divsqrt_multi.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_divsqrt_multi.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_fma_multi.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_fma_multi.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_fma.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_fma.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_noncomp.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_noncomp.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_opgroup_block.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_opgroup_block.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_opgroup_fmt_slice.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_opgroup_fmt_slice.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_opgroup_multifmt_slice.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_opgroup_multifmt_slice.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_rounding.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_rounding.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_top.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpnew_top.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpu_div_sqrt_mvp/hdl/defs_div_sqrt_mvp.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpu_div_sqrt_mvp/hdl/defs_div_sqrt_mvp.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpu_div_sqrt_mvp/hdl/control_mvp.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpu_div_sqrt_mvp/hdl/control_mvp.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpu_div_sqrt_mvp/hdl/div_sqrt_top_mvp.sv](https://chipsalliance.github.io/sv-tests-results/third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpu_div_sqrt_mvp/hdl/div_sqrt_top_mvp.sv.html) [third_party/cores/ariane/vendor/openhwgroup/cvfpu/src/fpu_div_sqrt_mvp/hdl/iteration_div_sqr [... cut ...]
                                                ^~~~~~~~~~~~~~~~~~~
```

Taken from the current deployment: https://chipsalliance.github.io/sv-tests-results/?v=slang+ariane+ariane

The first warning is about an incorrect include directory, could you check if that happens for you too? This is something that would preferably be fixed (even if it's just a warning), the errors seem like they could be real errors or maybe some files are still missing (or in incorrect order?)

Do you see the same errors/warnings locally btw?

Moschn, 2024-07-08:
I saw that error first too, but I believe that is/was an issue with the commit of Ariane that was chosen. When I bumped Ariane to the current master (as done in #5809), this issue disappeared. 

Btw: what is the ideal way to chose commits for third-party cores in this repo? Do you want to pick releases or just any somewhat recent commit?

tgorochowik, 2024-07-08:
Ideally we'd always have the latest versions of the cores, I missed the PR that bumps this. Let's merge it!

MikePopoloski, 2024-07-08:
For your earlier question, the LRM says this:
```
The syntax to declare a string variable is as follows:
string variable_name [ = initial_value ] ;
where variable_name is a valid identifier and the optional initial_value can be a string literal, the
value "" for an empty string, or a string data type expression.
```
and also:
```
Values of integral type can be assigned to a string variable, but require a cast. 
```
So the code is not legal. slang has a compatibility flag for this, which is included in "vcs" compatibility mode which I think sv-tests has turned on for Ariane so it should be ok (if not we can just add it if you don't want to change the code upstream).

The missing package error is real (slang doesn't care about the ordering of package files on the command line, though other tools may). It sounds like you've figured out the cause for this though.

Moschn, 2024-07-08:
Thanks for the detailed response!

I created a PR in upstream Ariane for this fix (https://github.com/openhwgroup/cva6/pull/2336).

Moschn, 2024-07-17:
The sv issue in ariane is fixed upstream. I am not sure why, but dependabot seems not to look for updates upstream. Otherwhise, I believe the ariane test is fixed and this can be closed.

Thanks for the help!

MikePopoloski, 2024-07-30:
Looks like the dependabot issues are resolved and the test is passing now.

PR #5808 (2024-07-08): Fix ariane generator (again)
Ariane seems to require recursive filelists. This commit adds recursive filelist parsing to the Ariane generator.

I am trying to set up a local instance of sv-tests, but compiling all the runners takes a while. In the meantime, we can also check what the CI says.

This commit (hopefully) fixes #5807
commit 49f1faa8 (2024-07-08): Fix ariane generator

Ariane seems to require recursive filelists. This commit adds recursive filelist parsing to the Ariane generator.

Fixes #5807
closed 2024-07-30 by MikePopoloski