chipsalliance/sv-tests #6511: Add "vast" library to comparison

parker-research, 2024-11-28:
I think it would be awesome to add the [vast library](https://github.com/vegaluisjose/vast) to this comparison. 

May come back and do it at some point, but just wanted to poke the discussion here.

wsnyder, 2024-12-13:
Looks like that project hasn't been maintained in 2 years, so probably not active enough?

parker-research, 2024-12-14:
Yeah, I guess so. Would still be nice to know where it stands in comparison to the rest. 

In theory, one of the selling points of Rust is that it's intended to be well-supported into the future with no maintenance, especially for smaller libraries (maybe not as much this one, but still curious to see how it performs in a benchmark).