chipsalliance/sv-tests #6973: Builds broken

wsnyder, 2025-02-25:
The dependabot updates have been broken since ~ Jan 6, so all data is a month stale.  Might this get fixed?

Or at least the verilator version updated if there's something hard to fix elsewhere, as I'd like to get back to regular checks that Verilator updates are not breaking any old passing tests.  (Every few months sv-tests would find something that regressed - so well appreciate being able to check!)

Thanks

@tgorochowik  et al.

tgorochowik, 2025-02-25:
Thank you for the report! We'll take a look

wsnyder, 2025-04-26:
Ping.  For some reason last dependabot build is here:

https://github.com/chipsalliance/sv-tests/pull/7022

I don't see any more recent for Verilator though there are regular changes.  Thanks for investigating.

tgorochowik, 2025-06-11:
Thank you for the reminders, unfortunately dependabot is not open source so it's hard to fully grasp what is happening, it does create PRs for other submodules but for whatever reason refuses to bump Verilator without reporting any issues or errors.

We'll investigate it once again and if nothing helps then maybe we'll add verilator under a different path to see if dependabot picks it up again?

MikePopoloski, 2025-06-13:
Looks like it's not just some dependabot pulls. The whole CI action seems busted for about a week now, due to "The job was not acquired by Runner of type self-hosted even after multiple attempts"

Definitely needs someone to spend a little bit of time on maintenance here. The test suite is nice, it would be a shame to just let it rot away.

tgorochowik, 2025-06-14:
The CI issue is a separate thing and something new, the infra team is looking into this

tgorochowik, 2025-06-25:
Dependabot works again, see #7240 and other RPs.

Apparently there is a bug in dependabot which makes it miscalculate the number of open PRs and prevents it from opening new PRs, this seems to have helped: https://github.com/chipsalliance/sv-tests/commit/41acf52030b5d43876e9c722f7599cd76801415a